nefarius/ScpToolkit #920: keep getting bsod when connecting ps4 controller

kewlp, 2017-09-24:
I keep getting scpvbus.sys bsod when connecting.
tried clean wipe already and manual wipe (but i can't find it in services.msc).

how do i fix this and stop getting BSOD?

Gnoce, 2017-09-25:
I have the same problem! Everything worked fine to this day. I tried everything but PC crash on BSOD as I switch on the DS4.
This is caused by the following module: scpvbus.sys
Bugcheck code: 0xC9 (0x7, 0xFFFFF803140C1B8C, 0xFFFF898D18722EA0, 0x0)
Error: DRIVER_VERIFIER_IOMANAGER_VIOLATION
file path: C: \ WINDOWS \ system32 \ drivers \ scpvbus.sys
product: Scp Virtual Bus Driver
company: Scarlet.Crush Productions
description: Scp Virtual Bus Driver
It's like Windows no longer recognizes the validity of this driver. Is there a solution or did they find a way to sell the xbox controllers?

Gnoce, 2017-09-26:
Ok, I found a solution:

1. Use ScpToolkit Clean Wipe Utility - Reboot
2. Re-install driver
3. Open command prompt as administrator
4. run verifier
5. select Delete existing settings.
6. Reboot the computer.
Done ;)

commit c082de82 (2018-01-26): ALL YOUR CODE ARE BELONG TO US! fixes #1039, fixes #1038, fixes #1037, fixes #1036, fixes #1035, fixes #1034, fixes #1033, fixes #1032, fixes #1031, fixes #1030, fixes #1029, fixes #1028, fixes #1027, fixes #1026, fixes #1025, fixes #1024, fixes #1023, fixes #1022, fixes #1021, fixes #1019, fixes #1018, fixes #1017, fixes #1016, fixes #1015, fixes #1014, fixes #1013, fixes #1011, fixes #1010, fixes #1009, fixes #1008, fixes #1007, fixes #1006, fixes #1003, fixes #1001, fixes #1000, fixes #999, fixes #998, fixes #997, fixes #994, fixes #993, fixes #992, fixes #991, fixes #990, fixes #989, fixes [... cut]
closed 2018-01-26 by nefarius