nefarius/ScpToolkit #922: False "fake controller" report

Nounours2627, 2017-09-25:
I've bought a new PS3 in a shop. I bought a seperate controller (to have two controllers) at the same moment in the same shop. 

I don't know which one is which one but the 2 controllers are CECHZC2E A1 and B1. Thet're not like my friend's controllers which are just ""CECHZC2E" nothing more. My friend's one is "Genuine" for SCP but my B1 is "fake" for SCP...

Does REAL shop sell fake controllers??? (and full price!) If not... SCP is probably considering the newest pad as fake.

rkubes, 2017-11-26:
I also have this issue with CECHZC2U B1; This is a "DUALSHOCK 3" "SIXAXIS" controller. I bought it at Wal-Mart. SCP claims it's fake, but there is no issue using it. The only "weird" thing is when the controller disconnects from Bluetooth due to the timeout, SCP pops up an error. This does not happen if I disconnect the controller by holding the PS button - only if it times out from inactivity. With that said, the error has no adverse effect. I can reconnect without issue.

rkubes, 2018-01-02:
I found the solution in #179. See the post from nefarius on Apr 18, 2016.

Quoted below for reference.

After you make this change, you need to restart the "SCP DSx Service" windows service (or restart Windows).

This will also supposedly fix any jitter that would occur when connected over USB. I haven't confirmed this yet.

> Your controller uses a Bluetooth chip from AzureWave Technologies Inc., did a bit of research and it looks like Sony also uses their chips in some VAIO notebook models so I guess we can assume the controller is genuine. Meanwhile you could try to add the line GenuineMacAddress = 08:A9:5A in ScpControl.ini below the section [Host Controller Interface] (on a "problematic" version ofc.) and see if behavior changes for you.

commit c082de82 (2018-01-26): ALL YOUR CODE ARE BELONG TO US! fixes #1039, fixes #1038, fixes #1037, fixes #1036, fixes #1035, fixes #1034, fixes #1033, fixes #1032, fixes #1031, fixes #1030, fixes #1029, fixes #1028, fixes #1027, fixes #1026, fixes #1025, fixes #1024, fixes #1023, fixes #1022, fixes #1021, fixes #1019, fixes #1018, fixes #1017, fixes #1016, fixes #1015, fixes #1014, fixes #1013, fixes #1011, fixes #1010, fixes #1009, fixes #1008, fixes #1007, fixes #1006, fixes #1003, fixes #1001, fixes #1000, fixes #999, fixes #998, fixes #997, fixes #994, fixes #993, fixes #992, fixes #991, fixes #990, fixes #989, fixes [... cut]
closed 2018-01-26 by nefarius