nefarius/ScpToolkit #927: Pressure sensitivity not working, stacktrace popup in pcsx2

TheRealMarnes, 2017-09-29:
I've got pcsx2 1.4.0, scptoolkit v1.7.277.16103-BETA, a dualshock 3 controller plugged into usb with a cable, and windows 10

I've installed scp according to instructions and whatnot, the controller works. It shows as a 360 controller, the joysticks and buttons all respond to input as they should, except that pressure sensitivity does not work.

I used the scp settings manager to patch pcsx2. When I started pcsx2, it got stuck loading padpokopom.dll, so I removed that file and tried again. Pcsx2 succesfully starts and recognizes the patched lilypad dll, and the controller works with it (without pressure sensitivity).

However, every time I start pcsx2 or open the lilypad settings and such, I get a stacktrace popup from scp:
[image]

I read in another issue that someone fixed it by uninstalling geforce experience. I did so and rebooted, error is still there.

How can I get this fixed and get pressure sensitivity to work?

TheRealMarnes, 2017-09-29:
I replaced my xinput.dll with the one from this issue: https://github.com/nefarius/ScpToolkit/issues/499

The stacktrace is gone but the buttons are still not pressure sensitive.

TheRealMarnes, 2017-09-29:
Aha, found the missing link! I didn't know this page existed: https://github.com/nefarius/ScpToolkit/wiki/Pressure-sensitivity-in-PCSX2

Ticking the ds3 native mode checkbox was all I needed to do

closed 2017-09-29 by TheRealMarnes

michaelxfung, 2017-10-20:
I got a same problem!
[image]
so... I renamed padPokopom.dll to padPokopom.bak skip the process.
[image]
just I setup a controller plugin, that keep looping show the error msg.