nefarius/ScpToolkit #929: First Controller connected to pc doesnt work ( via bluetooth )

deku55, 2017-10-02:
so basically the problem is that the first controller (DS3) connected to my pc doesnt work like it connects the lights turn on but no buttons work.  and when i connect my other controllers ( i have 4  all ds3)  they work completly fine. i tried connecting another controller first after a reboot and the same problem occurs. so the controllers work just fine 
i need help  wanna play 4 player games :(

deku55, 2017-10-07:
and btw the 2nd controller connected works as the first controller

Rudelic, 2017-10-21:
Have you checked if it receives an input from the remote?

kr4ft, 2017-10-31:
I think I've got the same bug here. I've got two DS3 controllers paired (via bluetooth). Also use a xbox one controller and a steam controller on the same PC.
Of the DS3 controllers, _only_ the second one works. Doesn't matter which of the two DS3 controllers is set to 2. 

Both DS3 controllers produce inputs which are detected in the Input Tester. However only the DS3 controller set to 2. Furthermore there is only one Xbox 360 controller installed in device manager.

deku55, 2017-11-13:
ok i fixed the problem all 4 controllers work now.

the problem was that under device manager there were 2 xbox 360 driver.
i deinstalled the one under "usb-devices" and then the controller started working.
the one that should remain is under "xbox 360  peripherals".
hope i could help some ppl with the same problem :)

commit c082de82 (2018-01-26): ALL YOUR CODE ARE BELONG TO US! fixes #1039, fixes #1038, fixes #1037, fixes #1036, fixes #1035, fixes #1034, fixes #1033, fixes #1032, fixes #1031, fixes #1030, fixes #1029, fixes #1028, fixes #1027, fixes #1026, fixes #1025, fixes #1024, fixes #1023, fixes #1022, fixes #1021, fixes #1019, fixes #1018, fixes #1017, fixes #1016, fixes #1015, fixes #1014, fixes #1013, fixes #1011, fixes #1010, fixes #1009, fixes #1008, fixes #1007, fixes #1006, fixes #1003, fixes #1001, fixes #1000, fixes #999, fixes #998, fixes #997, fixes #994, fixes #993, fixes #992, fixes #991, fixes #990, fixes #989, fixes [... cut]
closed 2018-01-26 by nefarius