nefarius/ScpToolkit #930: Windows 10 won't detect DS3 Controller (Device Descriptor Request Failed)

Danilux, 2017-10-04:
Hi, I have been going crazy, Windows 10 does not recognize my DS3 controller as a Playstation 3 controller, it shows "Unknown USB Device (Device Descriptor Request Failed)" every time I plugg it in, so I can't install any of the softwares that make PS3 controllers work like MotionInJoy or SCPToolkit because these software can't show my controller in order to apply the custom drivers on, has anyone run into this?

Danilux, 2017-10-11:
Ok, I have been making progress, I used for the first time the latest ScpToolkit v1.7.277.16103-BETA and the first controller now works, even wireless with BT dongle, but the second keeps giving me the device descriptor error, If I rerun the installer, it only shows the controller already working, not sure what to do to try en fix this, any ideas?

Danilux, 2017-10-12:
Here's some screenshot of the problem with the second controller, non of them is original though, maybe they both have the same hardware id? so they al clashing on windows and only one is being recognized?
[image]
[image]
[image]

Rudelic, 2017-10-21:
Seems like windows doesn't like your USB cable or the port you're connecting to, try to change to a different port, or change the cable.

Danilux, 2017-11-10:
One joystick works, the other one doesn't, I believe that joystick is simply bad, they are new but not original so, I'm pretty much hopeless at this point

commit c082de82 (2018-01-26): ALL YOUR CODE ARE BELONG TO US! fixes #1039, fixes #1038, fixes #1037, fixes #1036, fixes #1035, fixes #1034, fixes #1033, fixes #1032, fixes #1031, fixes #1030, fixes #1029, fixes #1028, fixes #1027, fixes #1026, fixes #1025, fixes #1024, fixes #1023, fixes #1022, fixes #1021, fixes #1019, fixes #1018, fixes #1017, fixes #1016, fixes #1015, fixes #1014, fixes #1013, fixes #1011, fixes #1010, fixes #1009, fixes #1008, fixes #1007, fixes #1006, fixes #1003, fixes #1001, fixes #1000, fixes #999, fixes #998, fixes #997, fixes #994, fixes #993, fixes #992, fixes #991, fixes #990, fixes #989, fixes [... cut]
closed 2018-01-26 by nefarius