nefarius/ScpToolkit #931: Built-in Blutooth not detected

abdalmoniem, 2017-10-05:
I have the Intel(R) Dual Band Wireless-AC 3165 bluetooth + wifi chip in my laptop and the scp driver installer doesn't detect it

Rudelic, 2017-10-21:
SCP makes bluetooth function by overwriting the driver necessary for the bluetooth dongle, effectively changing the whole way it performs to ONLY work with connecting PS3 controllers. SCP won't detect a combination chip like that because it only queries for bluetooth only adapters. Even if it did find your chip, you'd be left without WiFi, so I'd recommend getting a separate bluetooth dongle with bluetooth 4.0

matthijs186, 2017-11-26:
As far as I know, those Dual Band Wireless-AC devices have seperate Bluetooth and Wi-Fi drivers, and the latter shouldn't be affected if the Bluetooth driver is overridden (Wi-Fi works fine without the Bluetooth driver).
I'm not sure if that specific chip is supported, however. The README states that at least Bluetooth version 2.0 (2.1 for DS4 controllers) and EDR (Enhanced Data Rate) functionality must be supported.

abdalmoniem, 2017-11-27:
@matthijs186 yes I found a separate driver called `Zadig` that replaces the Bluetooth driver with an SCP compatible driver, my chipset is now detected in the SCP driver installation.

I found this solution [here](https://github.com/nefarius/ScpToolkit/issues/648)

_**I solved the problem I was having, now both DS3 and DS4 connect via bluetooth with SCPToolkit
I downloaded the SCP-DS-Driver-package-1.2.0.160 from this website:
http://forums.pcsx2.net/Thread-XInput-Wrapper-for-DS3-and-Play-com-USB-Dual-DS2-Controller**_

_**once downloaded go into the ../bin/Zadig directory
opened the Zadig program, click device>load preset, load the "bluetooth" preset
then click options>list all devices, select your bluetooth adapter from the list and click "replace driver"
afterwards load the DualShock4/DualShock3 preset (whichever you want to use), go to options>list all devices, choose the controller and click "replace driver"**_

_**I hope  this helps  someone else who had my problem :D**_

_Sent from my Galaxy Note4 using [FastHub-Libre](https://github.com/thermatk/FastHub-Libre/)_

closed 2017-11-27 by abdalmoniem