nefarius/ScpToolkit #935: No Bluetooth - Couldn't get connection handle [48, 20]

shua586, 2017-10-07:
Thank you for this project. It was working great for a little while but then all of a sudden the wireless capability stopped functioning. Whenever I disconnect my genuine PS3 controller from USB, it attempts to connect to Bluetooth but SCPMonitor pops an error that states "Couldn't get connection handle [48, 20]". 

I've tried 3 different dongles, and uninstalled/installed the drivers many times. Any help would be greatly appreciated! Thank you!

amcsi, 2017-12-17:
I'm getting the same issue and would also like to know how to fix this.

shua586, 2017-12-17:
I have no idea how I resolved this but eventually did. It started working
after uninstalling the Bluetooth drivers and reinstalling - all through the
SCPToolkit app

commit c082de82 (2018-01-26): ALL YOUR CODE ARE BELONG TO US! fixes #1039, fixes #1038, fixes #1037, fixes #1036, fixes #1035, fixes #1034, fixes #1033, fixes #1032, fixes #1031, fixes #1030, fixes #1029, fixes #1028, fixes #1027, fixes #1026, fixes #1025, fixes #1024, fixes #1023, fixes #1022, fixes #1021, fixes #1019, fixes #1018, fixes #1017, fixes #1016, fixes #1015, fixes #1014, fixes #1013, fixes #1011, fixes #1010, fixes #1009, fixes #1008, fixes #1007, fixes #1006, fixes #1003, fixes #1001, fixes #1000, fixes #999, fixes #998, fixes #997, fixes #994, fixes #993, fixes #992, fixes #991, fixes #990, fixes #989, fixes [... cut]
closed 2018-01-26 by nefarius