nefarius/ScpToolkit #939: Bluetooth HCI and LMP Version change after shutdown

manteero, 2017-10-09:
Thank you so much for making this, it helps me with connecting my DS3 to Laptop.

I'm using built in bluetooth (USB\VID_13D3&PID_3408&REV_0001)
when installing using SCP Driver installer, bluetooth and DS3 connect perfectly,
even after restart it still working (HCI Version: 6.3101, LMP Version: 6.0001)
but when after shutdown the HCI and LMP changed to (HCI Version: 7.3101, LMP Version: 7.0001)
and DS3 can't connect via bluetooth (blinking), while mac for both DS3 and bluetooth still same.

what can I do to fix this? or to make the HCI and LMP version stay/persistent at (HCI Version: 6.3101, LMP Version: 6.0001)??

ps: after changed to (HCI Version: 7.3101, LMP Version: 7.0001), even after several shutdown it stay the same HCI and LMP version

SilentPrayeCG, 2017-10-11:
I Guess i have same issue. I used Bluetooth dongle and it worked fine. But then i changed it with pciex card with wi-fi+bluetooth (but bt connected by usb to mother board, it is not pciex as i hoped)
This card is "Desktop wifi WLAN Atheros AR5B22 300M Wireless Wifi Bluetooth 4.0 PCI-E Card Desktop Adapter 6DB antenna" (name copied from ali)
BT from this card works fine, but, after sleep (i don't know but may be after shutdown too, i rarely shutdown pc)  or hibernation, gamepad stop connecting with BT, LEDs blinking, but nothing happens.
I tried, reboot, disable enable stuff and reset hubs (and even replug usb cord to mb). I'm check off "let's save power" flags. Do nothing.
Uninstalling bluetooth host (scp toolkit) bring back default bluetooth driver but installing scp bt back with driver installing utility doesn't help.

I guess F_USB2  and F_USB1, both daisy-chained, because looking at "power" tab in dev. mgr. i can see that one of hubs have something connected, (like F_USB1 have plugged F_USB2 in it or something)
Well, connecting BT in 2 or 1 seems doesn't make any difference, but probably if i plug BT in "leading" usb , and some flashdrive in "leader" (if i right about daisy-chain) it make that gamepad catch bt more often after sleep.
Or maybe i'm wrong and it just some kind of random.
Anyway only working way i found, it is use Clean Wipe utility first, and then reinstall scp drivers from scratch. After it gamepad catch usb again.

SilentPrayeCG, 2017-10-26:
Same thing happens when i stop SCP service (to use gamepad with my ps3), after i restarted it, HCI version became 7.1102 and gamepad stop catching bluetooth.

Andik86, 2017-11-12:
this is exactly same with my problem,

if i used **Qualcomm Atheros Bluetooth 4.0** (built in from laptop), works fine pairing 2 (maybe even 4) fake DS3 controller, until system reboot (i use windows 10 x64) HCI & LMP will changed to 7.xxxx

if i used **Bluetooth dongle USB v2.0**, i only can pair 1 controller via bluetooth (via USB works fine | related with with this issue ~ https://github.com/nefarius/ScpToolkit/issues/750), and HCI & LMP never changed even after system reboot.

have no clue which option i should use -_-

SilentPrayeCG, 2017-11-12:
For me it stops changing versions after sleep, i don't know what happen, it almost 2 weeks, i think, works fine. Don't know what changed. I tried unplug flash drive from pc case (it was plugged all time, but does not guarantee 100% issue free, because of it -it only theory that this do something with problem, probably not.)
I will watch what will be next.

Varon9, 2017-11-15:
same here atheros bluetooth 4.0 after a shutdown hci and mlp changed to 7.3101 &  7.0001
[image]

SilentPrayeCG, 2017-11-16:
Varon9 you still have creators update, try to update to fall creators. (1709 / 16299.64)
I don't sure, but it's only noticeable  thing that happens, before it stop changing versions for me.

PS
Today is also gamepad works ok, (it was in sleep and also rebooted once)

SilentPrayeCG, 2017-12-03:
After hibernation HCI version became 7.1xxx again... =(

commit c082de82 (2018-01-26): ALL YOUR CODE ARE BELONG TO US! fixes #1039, fixes #1038, fixes #1037, fixes #1036, fixes #1035, fixes #1034, fixes #1033, fixes #1032, fixes #1031, fixes #1030, fixes #1029, fixes #1028, fixes #1027, fixes #1026, fixes #1025, fixes #1024, fixes #1023, fixes #1022, fixes #1021, fixes #1019, fixes #1018, fixes #1017, fixes #1016, fixes #1015, fixes #1014, fixes #1013, fixes #1011, fixes #1010, fixes #1009, fixes #1008, fixes #1007, fixes #1006, fixes #1003, fixes #1001, fixes #1000, fixes #999, fixes #998, fixes #997, fixes #994, fixes #993, fixes #992, fixes #991, fixes #990, fixes #989, fixes [... cut]
closed 2018-01-26 by nefarius