nefarius/ScpToolkit #942: Constant input from R1/R2 and Triangle/Circle/Cross/Square

Arkann07, 2017-10-12:
Hello,

As the title says, my controller is acting like I'm constantly pressing R1/R2 and Triangle/Circle/Cross/Square, even though I'm not. Here is what the Input tester looks like:
[image]
My controller works fine on PS3, and used to work just fine on my previous computer. I tried using an older version of the wrapper and used other USB ports, without success. I am on Windows 10.

Any ideas whay may cause this ?

Rudelic, 2017-10-21:
Have you tried pressing the reset button on your remote? or replacing the cable which runs to your PC from the remote? Both of those have caused similar issues for me.

Arkann07, 2017-10-23:
Thank you very much Rudelic ! Luckily I had another cable so I tried both your suggestions, and now my controller works like a charm. Case closed !

closed 2017-10-23 by Arkann07