nefarius/ScpToolkit #950: Bluetooth stopped connecting pad

Uiasdnmb, 2017-10-17:
I had no issues running (latest non beta) version for last 6 months, but today Bluetooth just stopped working (fine over cable). Tried reinstalling this like 5 times (even beta version) to no avail

Anyone had a similar problem?
From what I see it connects but doesn't setup as a gamepad properly (screenshot taken after clearing logs and pressing ps button).

[image]

kosmic0, 2017-10-18:
same here sometimes when i disconnect the bluetooth the joypad connects and works fine but if i retsart pc same problem again the controller does not get detected an dkeeps flahsing

SilentPrayeCG, 2017-10-20:
You tried reinstall it through Clean Wipe + Driver install utility?

Uiasdnmb, 2017-10-20:
Multiple times but it didn't solve the problem (and caused some BSODs actually).
Ended up reconnecting dongle to another usb port and reinstalling, though it still doesn't work on "broken" port, so there's hoping it doesn't happen (again) enough times for me to run out of ports.

SilentPrayeCG, 2017-10-20:
You tried uninstall hub which contain this port?
I'm don't think some soft can break port physically.

Uiasdnmb, 2017-10-20:
Well it's not broken per se (hence the quotation marks), and bluetooth device did report pad being discovered, just never turned it into a viable controller.

I didn't want to uninstall the usb hub, since it's an internal hub and it's a pain to get into (my cables are a mess) and I have cpu pump connected there so I don't want to take risks of it having problems with reinstalling those.

SilentPrayeCG, 2017-10-20:
I meant just through device manager, to reinstall driver. But scp did not touch usb hubs drivers, as i know. It only overwrite default bluetooth driver.

commit c082de82 (2018-01-26): ALL YOUR CODE ARE BELONG TO US! fixes #1039, fixes #1038, fixes #1037, fixes #1036, fixes #1035, fixes #1034, fixes #1033, fixes #1032, fixes #1031, fixes #1030, fixes #1029, fixes #1028, fixes #1027, fixes #1026, fixes #1025, fixes #1024, fixes #1023, fixes #1022, fixes #1021, fixes #1019, fixes #1018, fixes #1017, fixes #1016, fixes #1015, fixes #1014, fixes #1013, fixes #1011, fixes #1010, fixes #1009, fixes #1008, fixes #1007, fixes #1006, fixes #1003, fixes #1001, fixes #1000, fixes #999, fixes #998, fixes #997, fixes #994, fixes #993, fixes #992, fixes #991, fixes #990, fixes #989, fixes [... cut]
closed 2018-01-26 by nefarius