nefarius/ScpToolkit #954: Vibration/Rumble doesn’t work on DS3 controller

punkdthrice, 2017-10-18:
I’m trying to get my DS3 controller to function properly on my PC. Originally I used the old Scarlett Crush Production drivers and vibration used to work. I did a reformat on my PC and it has never been the same.

Today I decided to install SCPToolKit for my Ps3 controller. All inputs and buttons work except for rumble/vibration. Disable rumble isn't checked in configuration nor is it disabled in game. Vibration works when controller is being used with my PS3 so the controller isn’t broken. Now it doesn’t work with XInput games. I haven’t tested with emulators.

How do I test rumble and get it to work properly? Should I run the Clean Wipe utility? I’m not sure what it does. Any help would be appreciated. Thanks in advance.

SilentPrayeCG, 2017-10-20:
Clean Wipe uninstall all scp related drivers and services, so you can install them through driver install utility again.
You can try it. There is warning, about risks, but since i have this problem https://github.com/nefarius/ScpToolkit/issues/939 i forced to use Clean Wipe very often, and so far it works without problems (and without reboots requiring).

It does't uninstall scp files from installation folder, only from system.

heltonteixeira, 2017-11-23:
Same issue here, tried different releases, did clean install a couple of times, tried to clean wipe, nothing made the force feedback/rumble work.

I'm using cable, and not the BT adapter, I don't know if it makes any difference.

commit c082de82 (2018-01-26): ALL YOUR CODE ARE BELONG TO US! fixes #1039, fixes #1038, fixes #1037, fixes #1036, fixes #1035, fixes #1034, fixes #1033, fixes #1032, fixes #1031, fixes #1030, fixes #1029, fixes #1028, fixes #1027, fixes #1026, fixes #1025, fixes #1024, fixes #1023, fixes #1022, fixes #1021, fixes #1019, fixes #1018, fixes #1017, fixes #1016, fixes #1015, fixes #1014, fixes #1013, fixes #1011, fixes #1010, fixes #1009, fixes #1008, fixes #1007, fixes #1006, fixes #1003, fixes #1001, fixes #1000, fixes #999, fixes #998, fixes #997, fixes #994, fixes #993, fixes #992, fixes #991, fixes #990, fixes #989, fixes [... cut]
closed 2018-01-26 by nefarius