nefarius/ScpToolkit #955: Pressure sensitivity can't working on PCSX2

michaelxfung, 2017-10-20:
I was setup pcsx2 1.2.1, scptoolkit v1.7.277.16103-BETA, a dualshock 2 controller plugged into usb adapter on windows 10.
I have a Blue DS4 controller plugged into usb cable, but scptoolkit can't detect it.
[image]

is holding on this configuration
[image]
so... I renamed padPokopom.dll to padPokopom.bak skip the process.
[image]
just I setup a controller plugin, that keep looping show the error msg.
so...I closed "ScpToolkit Tray Notifications", setup a controller plugin and go to "Test Device"
[image]
in PCSX2 my DS2 controller is no reaction.....
I try to setup on pcsx2 1.4.0, that is same result.

michaelxfung, 2017-11-18:
[https://github.com/nefarius/ScpToolkit/releases/tag/v1.6.238.16010](url)
New Version Fixed for this Problem, Thank you.
[image]
[image]
[image]

daygr, 2018-01-09:
Hmm, I'm not so sure. I get the same error when plugging in my controller, now. With ScpToolkit v1.7.277.16103-BETA.
[image]

commit c082de82 (2018-01-26): ALL YOUR CODE ARE BELONG TO US! fixes #1039, fixes #1038, fixes #1037, fixes #1036, fixes #1035, fixes #1034, fixes #1033, fixes #1032, fixes #1031, fixes #1030, fixes #1029, fixes #1028, fixes #1027, fixes #1026, fixes #1025, fixes #1024, fixes #1023, fixes #1022, fixes #1021, fixes #1019, fixes #1018, fixes #1017, fixes #1016, fixes #1015, fixes #1014, fixes #1013, fixes #1011, fixes #1010, fixes #1009, fixes #1008, fixes #1007, fixes #1006, fixes #1003, fixes #1001, fixes #1000, fixes #999, fixes #998, fixes #997, fixes #994, fixes #993, fixes #992, fixes #991, fixes #990, fixes #989, fixes [... cut]
closed 2018-01-26 by nefarius