nefarius/ScpToolkit #961: Please make option to disable popups

mikegonzalez2k, 2017-10-26:
My PS3 controller has a loose cable and as a result I have to unplug and replug it in every once in a while.
When I do lots of SCPToolkit popups appear which are distracting.
Please create a way to disable these popup notifications.
A better way might be to send them to a logfile so the user can read the errors.

ClintEstwood, 2017-11-02:
I have the same problem: a billion of Scp's messages.  
Please, add "Disable notifications" option.

Thanks!

phaylo, 2017-11-03:
The culprit here is "ScpTrayApp" process since it's the responsible for showing such notifications.
Killing "ScpTrayApp" process from Task Manager will get rid of the popups.

mikegonzalez2k, 2017-11-04:
Is the Tray app at all required for SCP to run properly though?  Does it do any sort of reinitialization of the app or maintain it's connectivity?

phaylo, 2017-11-05:
To be honest, I didn't take a look at the code base at all and only skimmed it. But I can tell you I've been using the controllers perfectly fine without the tray app for weeks.

commit c082de82 (2018-01-26): ALL YOUR CODE ARE BELONG TO US! fixes #1039, fixes #1038, fixes #1037, fixes #1036, fixes #1035, fixes #1034, fixes #1033, fixes #1032, fixes #1031, fixes #1030, fixes #1029, fixes #1028, fixes #1027, fixes #1026, fixes #1025, fixes #1024, fixes #1023, fixes #1022, fixes #1021, fixes #1019, fixes #1018, fixes #1017, fixes #1016, fixes #1015, fixes #1014, fixes #1013, fixes #1011, fixes #1010, fixes #1009, fixes #1008, fixes #1007, fixes #1006, fixes #1003, fixes #1001, fixes #1000, fixes #999, fixes #998, fixes #997, fixes #994, fixes #993, fixes #992, fixes #991, fixes #990, fixes #989, fixes [... cut]
closed 2018-01-26 by nefarius