nefarius/ScpToolkit #963: Vibration issue on win 10

Nani86, 2017-10-27:
Hello,
I've been using SCPtoolkit for my DS3 since I had win 7. I now have win 10. At first it was working fine even after the upgrade but a few months ago, my vibration function seemed to not be able to change amplitude. It still does seem like it can do different kinds of vibrations but it cannot do lower amplitudes anymore which kind of makes it pointless and difficult to use. I would greatly appreciate it if someone can provide a solution to this. Thank you.

axbmcuser, 2017-11-01:
Maybe try a clean install? I switched from win7 to win10 (clean install/no upgrade) recently, both systems got a fresh SCPtoolkit install and could not tell any difference while extensively testing original DS3 with vibration in many games/programs.

Nani86, 2017-11-02:
First of all, thanks a ton for the reply. 
By clean install,are you talking about the windows or SCPtoolkit? If you meant the first, that's near impossible for me at this point without investing 2 entire days and atleast an extra for backing up stuff and not to mention all the intricate customization I've done to my install. everything from codecs to registry edits etc. There are also plenty of software from various schools that I will loose the license for if I were to delete them XD. I will try reinstalling the SCP toolkit when I get a bit of time though :)

commit c082de82 (2018-01-26): ALL YOUR CODE ARE BELONG TO US! fixes #1039, fixes #1038, fixes #1037, fixes #1036, fixes #1035, fixes #1034, fixes #1033, fixes #1032, fixes #1031, fixes #1030, fixes #1029, fixes #1028, fixes #1027, fixes #1026, fixes #1025, fixes #1024, fixes #1023, fixes #1022, fixes #1021, fixes #1019, fixes #1018, fixes #1017, fixes #1016, fixes #1015, fixes #1014, fixes #1013, fixes #1011, fixes #1010, fixes #1009, fixes #1008, fixes #1007, fixes #1006, fixes #1003, fixes #1001, fixes #1000, fixes #999, fixes #998, fixes #997, fixes #994, fixes #993, fixes #992, fixes #991, fixes #990, fixes #989, fixes [... cut]
closed 2018-01-26 by nefarius