nefarius/ScpToolkit #969: DS4 weird USB behavior

SCP2K, 2017-11-06:
I've been trying to wrap my head around this for a couple of days with no success and searched for similar problems here to no avail. I hope you guys can help!

I've got 2 brand new PS4 controllers and installed scptoolkit. I has scarlet productions before for my DS3. Uninstalled it, and installed scp. I have repeated the installation + force install + clean wipe + manual driver uninstall etc. 

After installation, both my ds4 controllers seem to work fine while still connected to the usb cable - they are allocated a slot, they show as usb charging in the monitor. However, once I disconnect them and then reconnect they only charge (yellow blinking light) and cannot be used as wired controllers (cant turn them one, just charge them). They then appear in the monitor as usb 000000 none.
They also appear after reconnection under libusbK devices (which they don't directly after driver installation). My BT dongle also appears under libusbK.
During driver installation the ds4 controllers appear as wireless controller (interface 0) + interface 3 - two listings for each controller with different vids (one, for example: 054C). I tried installing just one interface or both for each controller and it did not seem to help.

The ds4 controllers seem to work fine via bluetooth after I disconnect or reconnect the usb cable for a while. However, one of them stops responding though appearing as connected after a few minutes (NBA 2k18, the other controller kept working. It's not a controller issue).

One last thing, which I'm not sure is related: after restarting the computer windows (7) seems to install some device. It seems to be my WIFI adapter, which has also been acting up since installing SCP. I made sure I did not select the adapter when installing the drivers. 

Any ideas on what's going on?
How to fix the usb\wired issue?
The BT connected yet not responding in a game after a few minutes?
The wifi issue?

Thanks to all in advance!

D1Vad22, 2017-12-08:
Hey Mate,

I had similar issue when connecting to usb and i fixed it!  I am Windows 10 and upgraded 1 of my controllers from PS3 to PS4. On USB cable it would show USB 000000 and another USB game controller would come up. This was annoying as if i was already in game and needed to charge it would come up as another device.

When controller was connected i went to Device Manager and uninstalled "Wireless Controller (Interface 0)" under "libusbK USB Devices" (deleted relevant files etc). I then installed driver for controller fresh through the SCP driver installer for PS4 controller and now showing as "Wireless Controller (Interface 3)" all working as expected via USB.

BUT the issue came back after reboot and next time i connected the controller via usb "Wireless Controller (Interface 0)" was back again. I stopped this happening by going to "SCPToolkit Settings Manager" > DualShock 4 specific > *uncheck* Always re-pair controller on Usb connection.

Just remember you need this setting *checked* in order to pair the controller to the PC first as you can't do the usual hold PS + Share button method with the bluetooth taken over by SCPtoolkit. So just enable this setting when pairing new PS4 controllers but always keep it unchecked :)

SCP2K, 2017-12-11:
Thanks!
I'll check this out soon.
What bothers me more, however, is the serious lag I've been experiencing,
which at times makes it feel as thought one of the two controllers isn't
working at all.
Do you have a fix for that? :)

Janni9009, 2017-12-13:
Normally for USB on the DS4, you shouldnt need any drivers on win7+. if you really need xinput, you can just use an Xbox360 controller emulator. other drivers might actually conflict with the controller. and if somehow got no dinput driver with windows, then steam should have one, and even a configuration assistant for the controller.

D1Vad22, 2017-12-13:
True, goodpoint. But i have 1x DS4 and 3x DS3. I need scptoolkit for the DS3 controllers. Not sure SCP2K has this requirement thougb.

SCP2K, 2017-12-17:
Thanks again.
I didn't know that Jani9009, thanks.
However, I am using SCP both for BT connectivity and because I want the
option of using DS4 + DS3 controllers, like D1Vad22.

So, my bigger problem is the BT connectivity thing and not the USB. I
mentioned the USB in case it was related to the other problem or a sign of
some faulty installation.
I should also mention that after installing (and reinstalling) the drivers,
by BT driver remains in the libusbK under device manager, I guess this
might be the problem?

commit c082de82 (2018-01-26): ALL YOUR CODE ARE BELONG TO US! fixes #1039, fixes #1038, fixes #1037, fixes #1036, fixes #1035, fixes #1034, fixes #1033, fixes #1032, fixes #1031, fixes #1030, fixes #1029, fixes #1028, fixes #1027, fixes #1026, fixes #1025, fixes #1024, fixes #1023, fixes #1022, fixes #1021, fixes #1019, fixes #1018, fixes #1017, fixes #1016, fixes #1015, fixes #1014, fixes #1013, fixes #1011, fixes #1010, fixes #1009, fixes #1008, fixes #1007, fixes #1006, fixes #1003, fixes #1001, fixes #1000, fixes #999, fixes #998, fixes #997, fixes #994, fixes #993, fixes #992, fixes #991, fixes #990, fixes #989, fixes [... cut]
closed 2018-01-26 by nefarius