nefarius/ScpToolkit #977: Scp Toolkit corrupted my laptop

SirSoundfont, 2017-11-16:
I installed Scp Toolkit a while ago, and used it for months without any problems. However, one time I tried using it, and I got an error message saying that windows found corrupt files, and a few seconds later, I got a BSOD. After my laptop restarted, I had permanently corrupted windows files, and I can't undo the damage that was done. Certain features don't work anymore at all, like my task manager. I've uninstalled it and removed it entirely, but things still aren't back to how they were before. (I know this was caused by Scp because the source of the corruption was "Scpvbus.sys")

matthijs186, 2017-11-26:
I have no idea if scpvbus.sys could do such harm to the system, and I cannot assist you there, but I can suggest the following:
Open a command prompt with administrator rights and run `chkdsk` to see if your hard drive is failing. If that is not the case, try `/sfc scannow` to see if Windows can recover all/most of the system files.

SirSoundfont, 2017-11-27:
I've tried that multiple times, and it said that it could not repair the damaged files. And my hard drive is failing.

JohnEdwa, 2017-11-30:
> And my hard drive is failing.

It's much more likely that your hard-drive failed first, and that caused the crash and BSOD than the other way around. The error saying the issue was with scpvbus.sys doesn't mean that it was the cause of the issue, only that it was the first that couldn't recover from it.

nefarius, 2018-01-16:
The `ScpVBus` doesn't interact with hard drives at all.

closed 2018-01-16 by nefarius