nefarius/ScpToolkit #978: This software is a worse than a bag of dicks

meffect, 2017-11-16:
This software is the worst trash I have run across in a long time. like 5 damn icons on my desktop after I install it. I have no idea what each of them do. all the apps look like early 10 years ago trash. The notifications make me want to gag. In the end I spent an hour trying to get it work, and the application led me on, to make me think it was working, but it wasnt. yeee yee

nefarius, 2017-11-16:
Wow, you are a terrible human being. Glad you remembered my why I disabled notifications.

closed 2017-11-16 by nefarius

meffect, 2017-11-16:
My philosophy is strong libertarianism. I'm one of the most ideal type of human that exists on planet earth.

Ryochan7, 2017-11-16:
I used to be a capitalist but one time I did too much DMT and the idea of separateness made no sense to me. ... You haven't done psychedelics so you don't f**king understand the nature of reality yet. You don't understand that we are all one.

Modern Libertarian mindset

meffect, 2017-11-17:
I wish I could do psychedelics ... however there's a giant ignorant wall in my way