nefarius/ScpToolkit #981: its 2017, and there's still this error

jxv16, 2017-11-21:
'Oh Sh... A fatal error occured. Consult the logs for details.'

how i fix this? T_T

rkubes, 2017-11-26:
Look through the logs and see if there's anything that stands out as an error.

They are usually located at: C:\Program Files\Nefarius Software Solutions\ScpToolkit\Logs

jxv16, 2017-11-27:
"<log4j:event logger="ScpDriverInstaller.MainWindow" timestamp="1511537227700" level="INFO" thread="1"><log4j:message>SCP Driver Installer 1.6.238.16010 [Built: 01/10/2016 09:52:11]</log4j:message><log4j:properties><log4j:data name="log4jmachinename" value="CIB8AEED00CEA9" /><log4j:data name="log4japp" value="ScpDriverInstaller.exe" /><log4j:data name="log4net:UserName" value="CIB8AEED00CEA9\alumno" /><log4j:data name="log4net:Identity" value="" /><log4j:data name="log4net:HostName" value="CIB8AEED00CEA9" /></log4j:properties><log4j:locationInfo class="ScpDriverInstaller.MainWindow" method="Window_Initialized" file="" line="0" /></log4j:event>
<log4j:event logger="ScpControl.Driver.NativeLibraryWrapper`1" timestamp="1511537227774" level="DEBUG" thread="1"><log4j:message>Preparing to load DIFxAPI</log4j:message><log4j:properties><log4j:data name="log4jmachinename" value="CIB8AEED00CEA9" /><log4j:data name="log4japp" value="ScpDriverInstaller.exe" /><log4j:data name="log4net:UserName" value="CIB8AEED00CEA9\alumno" /><log4j:data name="log4net:Identity" value="" /><log4j:data name="log4net:HostName" value="CIB8AEED00CEA9" /></log4j:properties><log4j:locationInfo class="ScpControl.Driver.NativeLibraryWrapper`1" method="LoadNativeLibrary" file="" line="0" /></log4j:event>
<log4j:event logger="ScpControl.Driver.NativeLibraryWrapper`1" timestamp="1511537227774" level="DEBUG" thread="1"><log4j:message>Called from 64-Bit process</log4j:message><log4j:properties><log4j:data name="log4jmachinename" value="CIB8AEED00CEA9" /><log4j:data name="log4japp" value="ScpDriverInstaller.exe" /><log4j:data name="log4net:UserName" value="CIB8AEED00CEA9\alumno" /><log4j:data name="log4net:Identity" value="" /><log4j:data name="log4net:HostName" value="CIB8AEED00CEA9" /></log4j:properties><log4j:locationInfo class="ScpControl.Driver.NativeLibraryWrapper`1" method="LoadNativeLibrary" file="" line="0" /></log4j:event>
<log4j:event logger="ScpControl.Driver.NativeLibraryWrapper`1" timestamp="1511537227774" level="DEBUG" thread="1"><log4j:message>DIFxAPI path: F:\ScpToolKit\DIFxApi\amd64\DIFxAPI.dll</log4j:message><log4j:properties><log4j:data name="log4jmachinename" value="CIB8AEED00CEA9" /><log4j:data name="log4japp" value="ScpDriverInstaller.exe" /><log4j:data name="log4net:UserName" value="CIB8AEED00CEA9\alumno" /><log4j:data name="log4net:Identity" value="" /><log4j:data name="log4net:HostName" value="CIB8AEED00CEA9" /></log4j:properties><log4j:locationInfo class="ScpControl.Driver.NativeLibraryWrapper`1" method="LoadNativeLibrary" file="" line="0" /></log4j:event>
<log4j:event logger="ScpControl.Driver.NativeLibraryWrapper`1" timestamp="1511537228010" level="DEBUG" thread="1"><log4j:message>Loaded library: F:\ScpToolKit\DIFxApi\amd64\DIFxAPI.dll</log4j:message><log4j:properties><log4j:data name="log4jmachinename" value="CIB8AEED00CEA9" /><log4j:data name="log4japp" value="ScpDriverInstaller.exe" /><log4j:data name="log4net:UserName" value="CIB8AEED00CEA9\alumno" /><log4j:data name="log4net:Identity" value="" /><log4j:data name="log4net:HostName" value="CIB8AEED00CEA9" /></log4j:properties><log4j:locationInfo class="ScpControl.Driver.NativeLibraryWrapper`1" method="LoadNativeLibrary" file="" line="0" /></log4j:event>
<log4j:event logger="ScpDriverInstaller.MainWindow" timestamp="1511537228013" level="INFO" thread="1"><log4j:message>Microsoft Windows 8.1 Professional (6.3.9600.0 AMD64) detected</log4j:message><log4j:properties><log4j:data name="log4jmachinename" value="CIB8AEED00CEA9" /><log4j:data name="log4japp" value="ScpDriverInstaller.exe" /><log4j:data name="log4net:UserName" value="CIB8AEED00CEA9\alumno" /><log4j:data name="log4net:Identity" value="" /><log4j:data name="log4net:HostName" value="CIB8AEED00CEA9" /></log4j:properties><log4j:locationInfo class="ScpDriverInstaller.MainWindow" method="Window_Initialized" file="" line="0" /></log4j:event>
<log4j:event logger="ScpControl.Driver.NativeLibraryWrapper`1" timestamp="1511537228016" level="DEBUG" thread="1"><log4j:message>Preparing to load libwdi</log4j:message><log4j:properties><log4j:data name="log4jmachinename" value="CIB8AEED00CEA9" /><log4j:data name="log4japp" value="ScpDriverInstaller.exe" /><log4j:data name="log4net:UserName" value="CIB8AEED00CEA9\alumno" /><log4j:data name="log4net:Identity" value="" /><log4j:data name="log4net:HostName" value="CIB8AEED00CEA9" /></log4j:properties><log4j:locationInfo class="ScpControl.Driver.NativeLibraryWrapper`1" method="LoadNativeLibrary" file="" line="0" /></log4j:event>
<log4j:event logger="ScpControl.Driver.NativeLibraryWrapper`1" timestamp="1511537228041" level="DEBUG" thread="1"><log4j:message>Called from 64-Bit process</log4j:message><log4j:properties><log4j:data name="log4jmachinename" value="CIB8AEED00CEA9" /><log4j:data name="log4japp" value="ScpDriverInstaller.exe" /><log4j:data name="log4net:UserName" value="CIB8AEED00CEA9\alumno" /><log4j:data name="log4net:Identity" value="" /><log4j:data name="log4net:HostName" value="CIB8AEED00CEA9" /></log4j:properties><log4j:locationInfo class="ScpControl.Driver.NativeLibraryWrapper`1" method="LoadNativeLibrary" file="" line="0" /></log4j:event>
<log4j:event logger="ScpControl.Driver.NativeLibraryWrapper`1" timestamp="1511537228043" level="DEBUG" thread="1"><log4j:message>libwdi path: F:\ScpToolKit\libwdi\amd64\libwdi.dll</log4j:message><log4j:properties><log4j:data name="log4jmachinename" value="CIB8AEED00CEA9" /><log4j:data name="log4japp" value="ScpDriverInstaller.e [... cut ...]
   en ScpDriverInstaller.App.Main()]]></log4j:message><log4j:properties><log4j:data name="log4jmachinename" value="CIB8AEED00CEA9" /><log4j:data name="log4japp" value="ScpDriverInstaller.exe" /><log4j:data name="log4net:UserName" value="CIB8AEED00CEA9\alumno" /><log4j:data name="log4net:Identity" value="" /><log4j:data name="log4net:HostName" value="CIB8AEED00CEA9" /></log4j:properties><log4j:locationInfo class="ScpDriverInstaller.App" method="&lt;OnStartup&gt;b__0" file="" line="0" /></log4j:event>"

i dont understand this T_T

Syst3mSh0ck, 2017-12-02:
It shows ScpDriverInstaller.exe crashed due to System.DllNotFoundException which is a dependency issue, possibly due to UAC. Logs show "Couldn't load library F:\ScpToolKit\libwdi\amd64\libwdi.dll" and looking up libwdi.dll confirms it to be a Windows Driver Installer library for USB devices which ScpDriverInstaller.exe uses to install the usb miniport class drivers for dualshock support. Make sure you right click on ScpDriverInstaller.exe and Run as Administrator otherwise it won't work properly.

jxv16, 2017-12-02:
i do that, and still not working T_T

commit c082de82 (2018-01-26): ALL YOUR CODE ARE BELONG TO US! fixes #1039, fixes #1038, fixes #1037, fixes #1036, fixes #1035, fixes #1034, fixes #1033, fixes #1032, fixes #1031, fixes #1030, fixes #1029, fixes #1028, fixes #1027, fixes #1026, fixes #1025, fixes #1024, fixes #1023, fixes #1022, fixes #1021, fixes #1019, fixes #1018, fixes #1017, fixes #1016, fixes #1015, fixes #1014, fixes #1013, fixes #1011, fixes #1010, fixes #1009, fixes #1008, fixes #1007, fixes #1006, fixes #1003, fixes #1001, fixes #1000, fixes #999, fixes #998, fixes #997, fixes #994, fixes #993, fixes #992, fixes #991, fixes #990, fixes #989, fixes [... cut]
closed 2018-01-26 by nefarius