nefarius/ScpToolkit #985: Wrong controller number DS3

HButlex, 2017-11-25:
Everytime i leave my controller and it goes to sleep the controller number changes to slot 2 and i can't change it without restarting my pc

danmakuz, 2017-11-26:
I'm also having this issue. Controller sleeps, I reconnect and it is assigned port 2

ryanssealey, 2017-11-28:
I'm having the same issue. I've tried switching around the "Always unplug virtual controller" and "Reserve controller slot" options in Service settings, but no luck. If the computer goes to sleep (I think that's when it happens), the controller gets assigned to Slot 2. The only way I've managed to get it back to Slot 1 is to restart the computer.

jayyyin, 2017-12-02:
I'm also getting this issue @nefarius , any workarounds you know of? it seems to think that a new controller is connecting when the previous connected one goes to sleep.

Master-Squishy, 2017-12-18:
I also get this, very annoying. Happens every time PC goes to sleep, can't seem to find an option anywhere to set controller to port 1 which makes no sense at all.

commit c082de82 (2018-01-26): ALL YOUR CODE ARE BELONG TO US! fixes #1039, fixes #1038, fixes #1037, fixes #1036, fixes #1035, fixes #1034, fixes #1033, fixes #1032, fixes #1031, fixes #1030, fixes #1029, fixes #1028, fixes #1027, fixes #1026, fixes #1025, fixes #1024, fixes #1023, fixes #1022, fixes #1021, fixes #1019, fixes #1018, fixes #1017, fixes #1016, fixes #1015, fixes #1014, fixes #1013, fixes #1011, fixes #1010, fixes #1009, fixes #1008, fixes #1007, fixes #1006, fixes #1003, fixes #1001, fixes #1000, fixes #999, fixes #998, fixes #997, fixes #994, fixes #993, fixes #992, fixes #991, fixes #990, fixes #989, fixes [... cut]
closed 2018-01-26 by nefarius