nefarius/ScpToolkit #989: ScpToolkit stops working after Sleep, Hibernate, and/or Signout.

doomrider7, 2017-12-03:
Exactly what it says on the topic header. Every time I put the computer on sleep, hibernate, or sign-out, I need to shut down and start my PC again to get to work. Any fixes for this issue?

danmakuz, 2017-12-07:
Make sure SCP service is running at boot up
Make sure there isn't a yellow warning in device manager

If it's the "unsigned driver" error that shows in device manager, it's because the driver for SCP is unsigned. There is a workaround for that.

Master-Squishy, 2017-12-28:
Hi I have this same issue, every single time it stops working I have to do a restart it gets really annoying, I leave the room to make a cup of tea come back and I have to shut down my PC start it up again and restart whatever game I was playing (witcher...) any way to get it to stop weirdly breaking on sleep

ToastyMisbehaviour, 2018-01-15:
Hello there, I have the exact same issue! I play Redout a lot and every time my computer goes to sleep, my game won't recognise any buttons anymore. I use a Ps3 controller. I tried everything from reinstalling the scp software to restarting the scp service in the service manager (which boots up with windows) and also restarting the game but none of this works. device manager says it works fine. Outside of the game, the controller works for some reason. It is so annoying to be forced to restart your computer every single time. Please help.

commit c082de82 (2018-01-26): ALL YOUR CODE ARE BELONG TO US! fixes #1039, fixes #1038, fixes #1037, fixes #1036, fixes #1035, fixes #1034, fixes #1033, fixes #1032, fixes #1031, fixes #1030, fixes #1029, fixes #1028, fixes #1027, fixes #1026, fixes #1025, fixes #1024, fixes #1023, fixes #1022, fixes #1021, fixes #1019, fixes #1018, fixes #1017, fixes #1016, fixes #1015, fixes #1014, fixes #1013, fixes #1011, fixes #1010, fixes #1009, fixes #1008, fixes #1007, fixes #1006, fixes #1003, fixes #1001, fixes #1000, fixes #999, fixes #998, fixes #997, fixes #994, fixes #993, fixes #992, fixes #991, fixes #990, fixes #989, fixes [... cut]
closed 2018-01-26 by nefarius