nefarius/ScpToolkit #994: controller won't connect to bluetooth

cruizerdog, 2017-12-06:
So there has probably been a ton of these comments made but i couldn't find a solution so submitting my own issue.

As title says i can get controller to work with cable but won't connect if i disconnect the cable as shown in the below pictures
[image]
[image]

Installed using scptoolkit setup.exe, installed the bluetooth driver and ps3 controller driver with genuine controller.
[image]
Assuming the bluetooth dongle was setup correcting as it is displayed as working with scptoolkit
[image]

i've done a few uninstalls and reinstalls of scptoolkit and the drivers and tried using SCP Pair tool suggested in a previously posted issue (however remote is already set as the host address and pressing set didn't seem to change anything anyway).

cheers for the replies in advance and sorry but i wasn't sure what debugging logs etc. i should have posted.

cruizerdog, 2017-12-06:
i also tried running SCPserver but it was doing the same as SCPmonitor/SCPservice and the pair button was greyed out

cruizerdog, 2017-12-06:
https://github.com/nefarius/ScpToolkit/wiki/Compatible-Bluetooth-Devices

So is this the only bluetooth adapters that work?

molitar, 2017-12-19:
The new version is total garbage!  The old version could force install on any Bluetooth device and it would work.  This new version is a total downgrade!  I am trying to figure out what version worked correctly as this 1.6 is crap!

commit c082de82 (2018-01-26): ALL YOUR CODE ARE BELONG TO US! fixes #1039, fixes #1038, fixes #1037, fixes #1036, fixes #1035, fixes #1034, fixes #1033, fixes #1032, fixes #1031, fixes #1030, fixes #1029, fixes #1028, fixes #1027, fixes #1026, fixes #1025, fixes #1024, fixes #1023, fixes #1022, fixes #1021, fixes #1019, fixes #1018, fixes #1017, fixes #1016, fixes #1015, fixes #1014, fixes #1013, fixes #1011, fixes #1010, fixes #1009, fixes #1008, fixes #1007, fixes #1006, fixes #1003, fixes #1001, fixes #1000, fixes #999, fixes #998, fixes #997, fixes #994, fixes #993, fixes #992, fixes #991, fixes #990, fixes #989, fixes [... cut]
closed 2018-01-26 by nefarius