nefarius/ScpToolkit #1002: Windows Defender User Interface bitcoin miner

alexaka1, 2017-12-16:
Hey guys!

So I just found a bitcoin miner on my computer. It was located in my AppData/Roaming/Nefarius Software Solutions folder, where there are two empty SCPToolkit folders and a folder named Windows Defender.

I have noticed increased temprature and GPU usage in my idle computer, after investigating, I found that this was the cause. MSASCui.exe with the old old brick wall Windows Defender icon, and it was eating up my GPU. I deleted the entire Windows Defender folder, and I am now running multiple virus scans to make sure there are no leftover registry edits or running processes.

Just wanted to let you know, I don't know if the github release got hacked or infected (or god forbid this is intentional on your part), but nevertheless I thought I'd let you know, because it was in your specific folder.

nefarius, 2017-12-16:
So what if it's in that folder? It's writable by any program running under your user. What version are you running? You got the setup from GitHub? My releases are signed, you checked that?

Innocence until proven guilty, same with malware so thanks for the report but I don't think I can help here 😏

alexaka1, 2017-12-16:
Yes I downloaded it straight from here, the GitHub repo.

I just didn't know how it got there, and since it wasn't in a new folder, but rather something associated with you, I came here to let you know. You never know these days. 😃 Also it could be some hacker decided to use your project for covering up their malware. 

And I didn't mean to accuse you of anything, I have used this toolkit before, though it was a much much earlier release, so I knew you guys wouldn't turn on us. I recently bought a new laptop and I downloaded this to it too. And a few days later the issue emerged. 

Thanks for the reply. It seems that I managed to remove it totally. I'll keep an eye out for it in the future.

closed 2017-12-16 by alexaka1

MarioMasta64, 2017-12-17:
nothing in my folder, sounds like an unlucky situation.