nefarius/ScpToolkit #1005: Can't install SCP

TheRealRef, 2017-12-19:
I'd been using SCP on my older PC running Windows 7.
On Friday I upgraded to a brand new PC w/ Ryzen R5 and a GTX1070 and _Windows 10_.
I download the package as zip, I run SCPUser.exe and I've been prompted always the same error "Error 0xc000007b". Tried everything, reinstall DX, VC 2010, VC2013, delete and redownload package. NOTHING.

On Windows 7 everything was smooth and clean, why can't I run it on Windows 10?

knives49, 2017-12-19:
SCPUser.exe ?
the zip is only source code if you want to compile yourself is good.
download the .exe version

TheRealRef, 2017-12-20:
OMFG. How could have I been so stupid? Downloaded the right version and worked fine. Sorry for useless post.

closed 2017-12-20 by TheRealRef