nefarius/ScpToolkit #1008: Error starting the service. Reinstall does not help.

justalemon, 2017-12-21:
## Error on the driver installer:
[image]
```
Couldn't start service:
System.InvalidOperationException: The service Ds3Service cannot be started on this computer
System.Component.Model.Win32Exception: The service or dependency group cannot be started
--- End of the procedure in the internal stack ---
  in
System.ServiceProcess.ServiceController.Start(String[] args)
  in
ScpDriverInstaller.MainWindow.StartService(String service)
```

## Error on services.msc:
[image]
```
Windows was not able to start the service SCP DSx Service on the Local computer.

Error 1068: The service or dependency group cannot be started.
```

KaneStaff, 2018-01-06:
I have the same issue. If I right click Ds3Service in Task Manager's Services tab and click Start, it says "The operation could not be completed. The dependency service or group failed to start." I installed the last stable one originally which worked for a long time. It probably stopped working with one of the recent Windows updates. I tried the latest beta version which also failed to work.

justalemon, 2018-01-08:
Yep, both the [stable](https://github.com/nefarius/ScpToolkit/releases/tag/v1.6.238.16010) and [beta](https://github.com/nefarius/ScpToolkit/releases/tag/v1.7.277.16103-BETA) shows the same error.

But using [Scarlet.Crush's](https://forums.pcsx2.net/Thread-XInput-Wrapper-for-DS3-and-Play-com-USB-Dual-DS2-Controller) version don't have the same problem so can be used as a workaround.

commit c082de82 (2018-01-26): ALL YOUR CODE ARE BELONG TO US! fixes #1039, fixes #1038, fixes #1037, fixes #1036, fixes #1035, fixes #1034, fixes #1033, fixes #1032, fixes #1031, fixes #1030, fixes #1029, fixes #1028, fixes #1027, fixes #1026, fixes #1025, fixes #1024, fixes #1023, fixes #1022, fixes #1021, fixes #1019, fixes #1018, fixes #1017, fixes #1016, fixes #1015, fixes #1014, fixes #1013, fixes #1011, fixes #1010, fixes #1009, fixes #1008, fixes #1007, fixes #1006, fixes #1003, fixes #1001, fixes #1000, fixes #999, fixes #998, fixes #997, fixes #994, fixes #993, fixes #992, fixes #991, fixes #990, fixes #989, fixes [... cut]
closed 2018-01-26 by nefarius