nefarius/ScpToolkit #1010: Service no longer coexists properly with genuine 360 controllers on Windows 10

sshurina, 2017-12-24:
I am running Windows 10 Professional x64, version 10.0.16299 Build 16299 updated to the most recent version as of this post. I am using SCP Toolkit version 1.6.238.16010

I am able to use multiple DS3 controllers together just fine. It also seems to work fine if you plug in 360 controller after all of the DS3 controllers. I have both "always unplug virtual controller" and "skip occupied gamepad slots" on, as well as the display controller number for the LED display. This problem only started happening recently, and I believe it may have happened after I updated windows (I defer updates so I have a better idea of when this may be related).

What seems to be happening is that it does not check for 360 controllers despite being configured to. So if a 360 controller is plugged in and is then followed by a DS3 controller, the DS3 controller ends up being assigned to have the same number as the 360 controller. This creates issues for the competing DS3 controller in terms of providing inputs -- windows prioritizes the real 360 controller in this instance.

It seems to properly skip slots for connecting the virtual bus, however -- both will show up in the device manager and other related services. The virtual bus is just never properly mapped to reference the DS3 controller, as upon viewing the raw inputs for the controller properties nothing I do seems to register. If I unplug the DS3 controller, then the virtual bus also stays there. If overlapping 360 and DS3 controllers are both disconnected and then the DS3 controller is plugged back in such that it lines up in order with the virtual bus (achieved with only DS3 controllers being plugged in), then it will disconnect properly once again. My guess is that it attempts to disconnect that of the same number as the overlapping 360 controller when they're both plugged in, which effectively does nothing, since something internally isn't lining up the DS3 controller up to the virtual bus properly.

If anyone knows of a fix to this, or at least some way to manually set what the controller number is for the DS3 controller using this tool, then that would be great.

Thanks

FeBauerschaefer, 2017-12-24:
Same Issue here need help asap.

sshurina, 2017-12-24:
It seems that SCPToolkit isn't being supported anymore, and instead the dev is making a new service for this due to issues in fixing bugs that would just be better solved by restructuring the programs.

https://vigem.org/  <- he has an announcement about it here on his site
https://github.com/nefarius/Shibari <- this is the new project

I'm not sure how well it works yet, but I figured I'd post this for anyone else having the same issue; I'm sure we'll have more luck with this long term.

commit c082de82 (2018-01-26): ALL YOUR CODE ARE BELONG TO US! fixes #1039, fixes #1038, fixes #1037, fixes #1036, fixes #1035, fixes #1034, fixes #1033, fixes #1032, fixes #1031, fixes #1030, fixes #1029, fixes #1028, fixes #1027, fixes #1026, fixes #1025, fixes #1024, fixes #1023, fixes #1022, fixes #1021, fixes #1019, fixes #1018, fixes #1017, fixes #1016, fixes #1015, fixes #1014, fixes #1013, fixes #1011, fixes #1010, fixes #1009, fixes #1008, fixes #1007, fixes #1006, fixes #1003, fixes #1001, fixes #1000, fixes #999, fixes #998, fixes #997, fixes #994, fixes #993, fixes #992, fixes #991, fixes #990, fixes #989, fixes [... cut]
closed 2018-01-26 by nefarius