nefarius/ScpToolkit #1019: Left Analog of PS3 Dualshock randomly pulls left

mikegonzalez2k, 2018-01-03:
While playing the left analog stick of my Dual Shock 3 controller randomly pulls to the left.  It functions perfectly for about 5 minutes.  Then some sort of event trigger causes it to thing the analog stick wants to be in the left position.  I have tried the controller on other devices such as my PS3 and my Galaxy Tab S3 tablet.  The controller works perfectly on those two devices.  It only seems to happen on Windows.  You may want to check if the driver has some sort of cycling that it loops through and checks and accidentally initializes this incorrectly.  Maybe some sort of centering issue with the left analog when it's cycled.  

Currently my only work around is to unplug the wired USB cable to make the controller re-initialize when I plug it back in.  This is why I believe it is something that happens during the cycling of the driver.  The weird thing is that sometimes it goes for a long period of time and other times it right away.  There doesn't seem to be a pattern to when it happens or an action which causes it.  It even occurs when I leave the controller sitting with no motion whatsoever.

OS: Windows 8.1 64 bit
SCPToolkit version: 1.6.238.16010

mikegonzalez2k, 2018-01-03:
Found a fix which I am testing.

I opened the SCPToolkit Settings Manager
In the Global Settings I set the Left Analog's Deadzone slider all the way to the right.
Since the issue was that it pulled left as it recentered, this seemed to instantly make the problem go away.  The controller seems to be perfectly centered.  I didn't even have to unplug the control to make it stop pulling.  I will post more if the problem persists.

rkubes, 2018-01-06:
When you connect your DS3, does it show up something along the lines as "Fake Dualshock 3" or not genuine? I forgot the exact text.
I had a similar issue with my genuine DS3 that was being incorrectly detected as fake. Once I corrected the issue so it recognized it as genuine, I no longer had the random upper-left pull.

If the MAC address of your controller starts with 08:A9:5A, then it could be the same issue. The solution is about half-way down in #179. Just search that thread for "scpcontrol.ini" to see how you need to modify that file. You'll have to restart the service afterwards (or restart Windows if it's easier for you).

I'm not sure why it matters whether SCP Toolkit detects the controller as genuine or not - maybe something wrong with the "workaround" code.

mikegonzalez2k, 2018-01-08:
Not sure what the MAC Address of the device is since I've never used it on my PC via bluetooth.  I always have connected via a mini USB charging cable to a USB port.  I can try to look at my DS3's MAC and see if it has the same address as the one you describe causes that problem.

mikegonzalez2k, 2018-01-08:
Well this old PS3 controller seems to not hold charge any longer.  Can't use it unless plugged in.

rkubes, 2018-01-09:
I haven't had a chance to play with this yet to determine actual steps, but you should be able to find the Mac address through USB. At worst I believe you can see it in the scp legacy monitor.

Regardless, it can't hurt just to follow those instructions to update the ScpControl.ini anyway. Best case it fixes it, worst case nothing changes.

mikegonzalez2k, 2018-01-10:
It looks like my DS3 is one of the ones with the MAC mentioned above.  I added that line to the ini, then restarted my PC just to make sure the drivers loaded properly.  So far so good, it's been 10 minutes and no issues yet.  Nefarious may want to add that MAC to future versions of the ini so other users don't get this same issue.  I'll let him know on Twitter.

mikegonzalez2k, 2018-01-10:
Well after letting it run for about 25 minute the problem came back :(

commit c082de82 (2018-01-26): ALL YOUR CODE ARE BELONG TO US! fixes #1039, fixes #1038, fixes #1037, fixes #1036, fixes #1035, fixes #1034, fixes #1033, fixes #1032, fixes #1031, fixes #1030, fixes #1029, fixes #1028, fixes #1027, fixes #1026, fixes #1025, fixes #1024, fixes #1023, fixes #1022, fixes #1021, fixes #1019, fixes #1018, fixes #1017, fixes #1016, fixes #1015, fixes #1014, fixes #1013, fixes #1011, fixes #1010, fixes #1009, fixes #1008, fixes #1007, fixes #1006, fixes #1003, fixes #1001, fixes #1000, fixes #999, fixes #998, fixes #997, fixes #994, fixes #993, fixes #992, fixes #991, fixes #990, fixes #989, fixes [... cut]
closed 2018-01-26 by nefarius