nefarius/ScpToolkit #1026: Install Issues & Service Issues - Need More Development Please....

sla0101, 2018-01-10:
Hello,

Is the development still active on this project?

I'm running Win7 SP1 x64 Home edition.

First problem, I got a popup saying disk1.cab had invalid digital signature.

Then I installed the Window Service, and I received a popup on the lower right of the desktop, saying it's already installed, like hmm, I didn't need this?

Then the SCP Dsx Service wouldn't start, something to do with a dependency issue, sorry I forget the exact wording.

Bottom line, I hope there will be some active development soon to get this up and working betting in Windows 7...

Thanks

sla0101, 2018-01-10:
Sorry my mistake, I didn't try hard enough, everything is working! :)

But I do hope that there will be more development in the near future.

Like there is no GUI to reassign the controller layout if someone wanted? 

Thanks

K9998f, 2018-01-11:
hi bro do you know how to change the quic disconnect combo?

commit c082de82 (2018-01-26): ALL YOUR CODE ARE BELONG TO US! fixes #1039, fixes #1038, fixes #1037, fixes #1036, fixes #1035, fixes #1034, fixes #1033, fixes #1032, fixes #1031, fixes #1030, fixes #1029, fixes #1028, fixes #1027, fixes #1026, fixes #1025, fixes #1024, fixes #1023, fixes #1022, fixes #1021, fixes #1019, fixes #1018, fixes #1017, fixes #1016, fixes #1015, fixes #1014, fixes #1013, fixes #1011, fixes #1010, fixes #1009, fixes #1008, fixes #1007, fixes #1006, fixes #1003, fixes #1001, fixes #1000, fixes #999, fixes #998, fixes #997, fixes #994, fixes #993, fixes #992, fixes #991, fixes #990, fixes #989, fixes [... cut]
closed 2018-01-26 by nefarius